LiquidPropulsionGroup/EngineWebServer #7: UART research

BalmaBrian, 2020-12-17:
I want to look into raspberry pie and Arduino communication through UART. This is the article I have on the topic so far https://roboticsbackend.com/raspberry-pi-arduino-serial-communication/

Maarten-Baert, 2020-12-18:
Hello,
If you can get your hands on a (digital) copy of "Make: AVR Programming" by Elliot Williams, that one is a good read in general in my opinion. 
It has a couple of chapters that go into detail about UART (Serial I/O) for the Atmel AVR microcontroller (used in Arduino but programmed in C in the book) and a lot of other things as well. 
Hope this is useful.

BalmaBrian, 2020-12-18:
@Maarten-Baert Thanks for the info, this will make this weekends programming a little more manageable.

BalmaBrian, 2020-12-28:
Terminal access through UART: [YouTube Link](https://www.youtube.com/watch?v=ZRKBlGvsxMw)
Basically what our REST server dose: [YouTube Link](https://www.youtube.com/watch?v=e38T-6C3mqk)

BalmaBrian, 2020-12-28:
This npm package might be what we go with for UART Serial Communication. [link](https://www.npmjs.com/package/serialport)

BalmaBrian, 2021-01-05:
We are going to throw this on the back burner.

closed 2021-01-05 by BalmaBrian