LiquidPropulsionGroup/EngineWebServer #8: Engine Web Server Wiki

BalmaBrian, 2020-12-26:
### Finish writing the first part of the wiki

Wiki Pages to be written:

- [x] Setting up the Raspberry Pi 
- [x] Installing Raspberry Pi OS
- [ ] Setting up SSH and optionally Wi-Fi
- [ ] SSHing into the Raspberry Pi
- [ ] Installing Updates
- [ ] Installing Environments

BalmaBrian, 2020-12-26:
[Setting up the Raspberry Pi](https://github.com/LiquidPropulsionGroup/EngineWebServer/wiki/Setting-up-the-Raspberry-Pi) done

BalmaBrian, 2020-12-26:
[Installing Raspberry Pi OS](https://github.com/LiquidPropulsionGroup/EngineWebServer/wiki/Installing-Raspberry-Pi-OS) done

chubbychow, 2020-12-26:
this is an article on how to connect via ssh using rasberry pi.   [https://www.raspberrypi.org/documentation/remote-access/ssh/windows10.md](url)

BalmaBrian, 2020-12-26:
img directory created in [this is the commit](335391744a4123be38cc510a349a5accdf8cfcb4). Screen shots and wiki pitures are put there for reference in page creation.

BalmaBrian, 2020-12-27:
Video made for wiki
[image]

BalmaBrian, 2020-12-27:
Steps for the `Setting up SSH and optionally Wi-Fi` page:

1. make ssh file somewhere (not boot on sd card)
2. make wpa_supplicant.json on boot on sd card
3. populate wpa_supplicant.json
4. power your board
5. locate your boards ip
6. ssh into your board
7. credentials for your pi

BalmaBrian, 2020-12-27:
screen cap converted into gif
[image]

BalmaBrian, 2020-12-28:
new screen cap
[image]

BalmaBrian, 2021-01-01:
[SSHing into the Raspberry](SSHing into the Raspberry Pi) Pi page done.

BalmaBrian, 2021-01-01:
[Setting up SSH and Optionally Wi-Fi](https://github.com/LiquidPropulsionGroup/EngineWebServer/wiki/Setting-up-SSH-and-optinally-Wi-Fi) Done.

BalmaBrian, 2021-01-01:
[Installing Environments](https://github.com/LiquidPropulsionGroup/EngineWebServer/wiki/Installing-Environments) Done.

BalmaBrian, 2021-01-01:
Final page [Enabling SSH Server](https://github.com/LiquidPropulsionGroup/EngineWebServer/wiki/Enabling-SSH-Server) Done.

closed 2021-01-01 by BalmaBrian