LiquidPropulsionGroup/EngineWebServer #14: Activity Diagram

BalmaBrian, 2021-01-08:
Simple Activity Diagram to show how v1.0 of the Engine Web Server works.

BalmaBrian, 2021-01-08:
Activity Diagram Transparent Background
[image]

BalmaBrian, 2021-01-08:
Activity Diagram Solid Background
[image]

BalmaBrian, 2021-01-08:
@M-LKT  @chubbychow What do yall think? Also the images should be posted on the Discord.

TWCCarlson, 2021-01-08:
Controller has 2 l's
Image embeds on GitHub seem to be broken?
Should GUI requests get a separate diagram?

BalmaBrian, 2021-01-08:
@TWCCarlson  
- This activity diagram dose not break down the UI because this repo is not the UI repo.
- Spelling errors fixed
- png's seem not to work well in issues

TWCCarlson, 2021-01-08:
Requests from the GUI to retrieve information are being handled by the webserver are they not?

BalmaBrian, 2021-01-13:
Basic Activity Diagram for Engine Web Server v1.0 done.

closed 2021-01-13 by BalmaBrian